breakintoprogram/agon-mos #15: Turn off LF/CRLF conversion in FatFS string functions

breakintoprogram, 2023-02-14:


breakintoprogram, 2023-02-14:
The only conversion I can see is here.
Line 6500 of ff.c
```
if (FF_USE_STRFUNC == 2 && c == '\n') {	 /* LF -> CRLF conversion */
	putc_bfd(pb, '\r');
}
```
Setting FF_USE_STRFUNC to 1 in ffconf.h will turn off LF -> CRLF conversion

envenomator, 2023-02-15:
I always forget which one is which (CR/LF). Just tested with a small piece of code, writing to a file with fa_write | fa_create_always. As soon as I write a 0x0a, the 0x0D is immediately added as well. When I write a 0x0D, just 0x0D gets written, nothing extra.

envenomator, 2023-02-15:
Defined FF_USE_STRFUNC as 1 in ffconf.h and reflashed MOS 1.02; no additional 0x0Ds are added anymore, which is good. Also checked if saves/loads in BBCBASIC still work, which they do. So I think indeed setting this solves the issue; great!

closed 2023-03-20 by breakintoprogram