breakintoprogram/agon-mos #48: USB Keyboard Support

fadeddata, 2023-04-11:
Just received an Agon Light from Olimex. It has the new USB port but it doesn’t actually support USB keyboards. It supports keyboards that speak PS2 protocol (over USB connector?) Unfortunately I don’t have a keyboard that does that. I do remember when keyboards were made like this. Basically they supported both protocols and you used a passive converter. Ideally MOS would support USB HID? Maybe there is another option?

leventp, 2023-04-11:
The following  Amazon keyboard works with Olimex Agon Light. 

Amazon Basics USB Wired Computer Keyboard and Wired Mouse Bundle Pack
https://www.amazon.com/dp/B00B7GV802?psc=1&ref=ppx_yo2ov_dt_b_product_details

breakintoprogram, 2023-04-12:
I'm going to mark this as 'Won't Do' as it is a hardware limitation of the platform that can't be resolved in software.

closed 2023-04-12 by breakintoprogram