breakintoprogram/agon-mos #55: RUN and JMP should push and pop registers

rprouse, 2023-04-25:
Currently assembly programs need to push all the registers before calling main then pop the registers before the program returns. Most other operating systems do this for you, so it is confusing for new developers. C programs also need to include assembly to do the same before calling into `main`.

Writing code like this in every program seems a waste. It would be much easier if MOS did it before and after running the program.

```asm
_start:
	push af
	push bc
	push de
	push ix
	push iy

	call _main

	pop iy
	pop ix
	pop de
	pop bc
	pop af

	ld hl, 0
	ret
```

Added bonus, also parse the arguments and put them in a defined location in memory? 😄

rprouse, 2023-04-25:
I've tested stripping out the `push` and `pop`  instructions and the program still runs although I didn't test running as a command. Are they needed?

If I remove the `ld hl,0`, then garbage is printed before the command prompt returns.

breakintoprogram, 2023-04-27:
HL is used as the MOS return error code. 0 = OK. There is no bounds checking on that table. I will look into that as part of this fix.

I'd argue that JMP doesn't need to push registers as it is not intended to return neatly to MOS, but will look into it for RUN.