breakintoprogram/agon-mos #57: UART baud rate appears to be off.

lennart-benschop, 2023-05-16:
In src/uart.h MASTERCLOCK is set to 20000000, which should be 18432000. The resulting UART baud rates are therefore 8% lower than specified. For the VDP link it makes no difference as the baud rate divisor in the eZ80 is rounded to 1 in any case.

lennart-benschop, 2023-05-16:
The function UART1open contains code like this, so thie wrong MASTERCLOCK appears to be used indeed.

	UINT32	mc = MASTERCLOCK;										// UART baud rate calculation
	UINT32	cb = CLOCK_DIVISOR_16 * pUART->baudRate;				// split to avoid eZ80 maths overflow error
	UINT32	br = mc / cb;											// with larger baud rate values

breakintoprogram, 2023-05-25:
Closing this as it is a duplicate of #58

closed 2023-05-25 by breakintoprogram