breakintoprogram/agon-mos #63: Can't make *SAVE work

davidgiven, 2023-05-27:
(This is all from BBC Basic.)

`*SAVE x 1000 1001` produces  a file which is 1001 (decimal) bytes long (I'd expect it to be 1).

`*SAVE x 1000 &1001` produces a file which is 4097 (decimal) bytes long (I'd expect it also be 1).

`*SAVE x 1000 +10` produces a file which is 10 (decimal) bytes long (likewise 16).

I also don't think it's saving data at the right address:

```
DIM A 1
?A=&55
PRINT A, ~A
```

...then saving one byte using the two addresses above, first with the decimal address and second with the hex address prefixed with `&`, produce files with different content, neither of which contains a &55. I've tried adding on &40000 (I think that's where BBC Basic loads? It's unclear) and that doesn't help.

<edit>

So I've found that the start/end semantics above does actually match the description in the Agon MOS documentation, even though it's different from Acorn MOS. It's also different from the syntax described in the Agon BBC Basic documentation, which matches Acorn's. So I don't know whether the change is accidental or not. I do suspect that there's no way in which the 16-bit addresses used in the BBC Basic documentation can work when passed to ADL MOS unless BBC Basic overrides *SAVE and *LOAD, but if that's the case then there needs to figure out the absolute address of a data block otherwise they can't work from Basic.

So I think there's still something wrong, but am not sure what. (Also I still can't save my data!)

breakintoprogram, 2023-05-29:
Hi, the documentation for *SAVE and *LOAD is here for MOS This is the command you are running from BBC BASIC
https://github.com/breakintoprogram/agon-docs/wiki/MOS#save
The third parameter is size, not end byte.
As it is a MOS command, the start byte is 24-bit, not 16-bit, so you need to add the base address of BBC BASIC which is &40000.

breakintoprogram, 2023-05-29:
Also thank you for setting me off on this journey btw.

davidgiven, 2023-05-29:
I think you should probably document that it takes _decimal_ numbers, not hex --- that tripped me up too (it only accepts hex with the & prefix). The number validation is a bit iffy, too, it would happily accept hex numbers and then not do the right thing with them. (It's possible this is why I couldn't make it work. It took me a while to figure out what was happening.)

breakintoprogram, 2023-05-30:
Actions:

- Improve documentation regarding number formats in MOS calls
- Add validation and appropriate error messages if invalid parameters passed

breakintoprogram, 2023-05-30:
Fix bug in mos_parseNumber as follows:

Line 255-258:

```
value = strtol(p, &e, base);
if(*e != 0) {
	return 0;
}
```
Fix mos_cmdRun to accept the default "." character

closed 2023-06-27 by breakintoprogram