breakintoprogram/agon-mos #65: Negative file sizes for large files

nihirash, 2023-05-30:
I've found it when used CP/M disk images(8 megs) - example files can be found here: https://github.com/nihirash/Agon-CPM2.2

If we ask system for directory listing we got negative file sizes.

jonathan-teh, 2023-06-10:
Seen this too and it affects `CAT` with files of 8MiB or more. This should fix it:
```diff
diff --git a/src/mos.c b/src/mos.c
index be27a05..f9b97d3 100644
--- a/src/mos.c
+++ b/src/mos.c
@@ -748,7 +748,7 @@ UINT24      mos_DIR(char * path) {
                        hr = (fno.ftime & 0xF800) >> 11;        // Bits 15 to 11
                        mi = (fno.ftime & 0x07E0) >>  5;        // Bits 10 to  5
                        
-                       printf("%04d/%02d/%02d\t%02d:%02d %c %*d %s\n\r", yr + 1980, mo, da, hr, mi, fno.fattrib & AM_DIR ? 'D' : ' ', 8, fno.fsize, fno.fname);
+                       printf("%04d/%02d/%02d\t%02d:%02d %c %*lu %s\n\r", yr + 1980, mo, da, hr, mi, fno.fattrib & AM_DIR ? 'D' : ' ', 8, fno.fsize, fno.fname);
                }
        }
        f_closedir(&dir);
```

breakintoprogram, 2023-07-08:
Thanks for the fix - much appreciated.

closed 2023-07-08 by breakintoprogram