breakintoprogram/agon-mos #68: MOS commands ending in numeric parameter not executed from autoexec.txt

lennart-benschop, 2023-06-03:
This started to go wrong since v1.04RC1.

autoexec.txt has lines ending in CR-LF. This is normal for text editors under Windows and also Nano on Agon itself.

When a command like "SET KEYBOARD 1" occurs in that file, it is not executed. If I put an additional space after the '1' it works again.

lennart-benschop, 2023-06-03:
These commands work when executed from the MOS command line itself, just not from autoexec.txt

breakintoprogram, 2023-06-06:
Please check you have pulled the correct src_fatfs/ffconf.h - FF_USE_STRFUNC should be 1. Dated 13/04/2023.

lennart-benschop, 2023-06-07:
I checked that I have you current src_fatfs/ffconf.h file with -FFUSE_STRFUNC set to 1.
All file system reads by user programs appear to work correctly, just the strange behaviour with lines in autoexec.txt file.

breakintoprogram, 2023-06-07:
It is quite possible that this is a bug - I've improved error checking on parameters and I can see why that may fail.

breakintoprogram, 2023-07-08:
The MOS now trims whitespace from any command executed by mos_exec.

closed 2023-07-08 by breakintoprogram