breakintoprogram/agon-mos #69: mos_fclose not usable if VDP in terminal mode

nihirash, 2023-06-05:
It tries fetch RTC data but have no possibility.

Please add flag to ignore RTC data - cause my CP/M implementation working in terminal mode and there no RTC data.

leighbb, 2023-06-12:
I don't the history around some of the design choices. Having said that, would it not be better to switch to using the ez80 RTC for this stuff?

leighbb, 2023-06-13:
I think there is still room for a "raw" Terminal approach, but I think it is a bit limiting. I would prefer an approach that allows you to use user defined terminal sequences to achieve the same functionality when operating in Terminal mode.  There may still be a part to play for a completely "raw" terminal mode, but I think an approach that still allows you to use the full MOS functionality would be better.

I'll share a rough outline of the approach when I get a chance. I have it more or less working, but need to tidy it up.

nihirash, 2023-06-13:
This patch mainly required for CP/M. In MOS mode there no requirements for this.

sijnstra, 2023-06-21:
Not just required to CP/M mode. I've converted my infocom interpreter to use terminal mode to have access to 80 columns and underline in the font, as well as other VT100 functionality. Now I can't save games as that required the datestamp for file creation.

nihirash, 2023-06-21:
As adhoc solution - you can grab MOS binary from cp/m chat in discord - it's vanilla MOS with my simple patch.

sijnstra, 2023-06-24:
I ended up compiling from your fork - I have to justify buying that zilog cable somehow! Anyway, confirmed that it's working and solved my issue with creating new files while in terminal mode. This patch allowed me to confidently release my infocom interpreter, however, I am absolutely happy to change my code if a different agreed method for getting around this issue turns up.

nihirash, 2023-06-25:
You can flash it just via https://github.com/envenomator/agon-flash

If something will go wrong - you can flash Agon HAL firmware for ESP with support uploading flash via USB and when computer starts successfully rollback to Quark

breakintoprogram, 2023-07-08:
Closing as 'Won't Do' as this is a duplicate of #70.

closed 2023-07-08 by breakintoprogram