breakintoprogram/agon-mos #76: Suggestion : add Jeroen Venema's patches to allow hex loading of files.

paulscottrobson, 2023-07-19:


breakintoprogram, 2023-08-03:
I'll have a think about this - I'm kind of 50/50. The original intention behind the MOS was for it to be a lightweight OS with extra functions being provided by external * commands. This kind of goes against that; adding functionality and complexity back in. If there is a significant argument for this, I'm all ears, otherwise will mark as 'Won't Do'.

envenomator, 2023-08-06:
I’m actually 50/50 as well for exactly this reason, which is why I never created a PR for this. The transfer over gpio uart is fine using as a command under /mos. The only reason for considering adopting it in mos could be that the patch, enabling transfer over the esp32 usb serial, could enhance the average physical life of the sd card slot for users cross-developing code. By having this functionally as standard, chances for people using it this way could increase versus them just not bothering to buy an ftdi interface and swapping sd cards. A stretch maybe.
As a command, the esp32 usb serial route will never work without vdp/esp32 cooperation.