breakintoprogram/agon-mos #78: Key inputs go missing after pressing caps lock

stevesims, 2023-08-11:
For a short period of time after pressing the caps lock key subsequent key presses will be ignored.

This can easily be reproduced either at the MOS prompt or a BASIC prompt.  Press the caps lock key and then immediately type a letter, and nothing will show up.

It feels like keyboard input takes about 1s to start working again.

theflynn49, 2023-08-11:
On my Olimex Agon light 2, VDP 1.04 RC1, MOS 1.04 RC1, basic 1.05 , keyboard from Olimex "delux", it works perfectly.

stevesims, 2023-08-11:
After I posted this I noticed that there has been some new commits made to this repo which appear to affect how the keyboard is handled.  I need to update my MOS and re-test, but that might take me a little while.

Similarly it seems I'm running version 1.05 of BASIC so I'll need to update that too.

At least my VPD is up to date 😀 

I'll try to update my MOS and BASIC soon and re-test

breakintoprogram, 2023-08-13:
Thanks. I'll mark this card as 'Won't Do'

closed 2023-08-13 by breakintoprogram