breakintoprogram/agon-mos #80: Micro SD Does not seem to be found, how to format micro SD?

ainslec, 2023-08-12:
I'm on MOS and VDP 1.02 of Agon Light Olimex edition.

I formatted a microsd card as FAT32, and put on bbcbasic.bin in the root of the card but I see a FAT error message.

What is the likely cause of the error do you think? Bad Agon hardware, faulty formatting?

How do you recommend I format a microsd card, and what capacity cards and sd types are supported?

[image]

[image]

breakintoprogram, 2023-08-13:
Hi, I'm not a Windows user, sorry, so can't offer any OS specific advise. Could you tell me the advertised capacity and class of the SD card and I may be able to advise further. Some older, slower SD cards may not be read correctly by the firmware.

ainslec, 2023-08-13:
Hello, please close this. I found the answer I was looking for in the Agon manual [here](https://github.com/TheByteAttic/AgonLight).

I was using a 2GB class 4 micro sd card, which I had assumed would not be a problem at all.

After trying with a 32GB class 10 card (total overkill), MOS worked. The previous card was formatted fine with FAT32, and I have no idea why Agon needs a class 10 card, but problem solved. Cheaper than a broken Agon. It'll take a while to fill 32KB, let along 32GB.

[image]

breakintoprogram, 2023-08-13:
Hi, will do, thanks for the update, and glad you got it working.

It's to do with the firmware timings more than anything. It * may * work on cards slower than Class 10, but we've only really been able to test it on a small subset.

closed 2023-08-13 by breakintoprogram