breakintoprogram/agon-mos #81: Build instructions missing step to get to a MOS.bin file

stevesims, 2023-08-13:
[Instructions for updating firmware](https://github.com/breakintoprogram/agon-docs/wiki/Updating-Firmware) describe the process of updating MOS using the agon-flash utility with a MOS.bin file.

The build instructions for agon-mos only talk about using ZDSII with a special cable.  Building the project with ZDSII does not produce a MOS.bin file.  I'm not familiar with ZDSII, so I'm unsure how to proceed from this point.

The instructions are missing a step for how to get from the output of ZDSII to a MOS.bin file that one can use with the agon-flash utility.

tonedef71, 2023-08-13:
> [Instructions for updating firmware](https://github.com/breakintoprogram/agon-docs/wiki/Updating-Firmware) describe the process of updating MOS using the agon-flash utility with a MOS.bin file.
>
> The [...]

Download the [HexToBin tool](https://sourceforge.net/projects/hex2bin/) and use it to create a .BIN file from the .HEX file that is produced by ZDSII (DEBUG profile).

stevesims, 2023-08-13:
Thanks for that @tonedef71 

Using HexToBin I did manage to get a .bin file, however when I try to install it my Agon does nothing.  The VDP hangs at "wait_eZ80: Start", so something clearly isn't quite right.

I tried using the agon-vdpflash to install the MOS.bin file I'd produced, to no avail.  I also tried the "release" profile MOS.hex file as a source, also without success.

I've gone back to the Jul 10th release of 1.04RC1, and it's booting again.

I'm using an M1Max Mac, and running ZDSII on ARM Windows in Parallels, plus I'm having to recompile hex2bin...  Using the pre-compiled Windows version of hex2bin inside Windows yields me a MOS.bin file with exactly the same CRC as the hex2bin version I'd built on Mac OS, so that variable can be ruled out.

My suspicion is that there is something wrong with my ZDSII.  When I check out the 1.4RC1 tag on this repo, build, and run hex2bin the resultant MOS.bin file I get has a different CRC than the 1.4RC1 MOS.bin release file I've downloaded. 😞

theflynn49, 2023-08-13:
LuzrBum has done a video on "Hacking the kernel" here www.youtube.com/watch?v=gt1HKQ27KRI&
He gives a few hints on which exact version of ZDSII you should use (Acclaim) and what you should do but I don't know if it's complete. I never recompiled MOS so I just don't know. He suggests to test your bin file with the emulator before risking to brick your Agon, even if it's said to be impossible. My personal feeling : if the bin file you generated with the 1.04 sources is different than the released one, you have a wrong library somewhere.

closed 2023-08-13 by breakintoprogram

stevesims, 2023-08-14:
I don't understand why this issue has been closed

Whilst I have been given some help that _might_ allow me to build this repo, the documentation issue remains.  Neither the build instructions in the Read Me file for this repo, nor the documentation wiki, have been updated to provide clearer instructions for how to produce a `MOS.bin` file from this repository.

stevesims, 2023-08-14:
Ah - apologies - I've just spotted the latest commit to this repo - the Read Me has indeed been updated.

Apparently I'm blind.  :grin: