breakintoprogram/agon-mos #100: Release needs to include/provide hash.

paulscottrobson, 2023-10-19:
I've just installed 1.04RC2, which seems to so far work fine. However to use Flash you need the CRC32 of the MOS.bin file. I calculated it with rhash, but this rather invalidates the point of verifying the file. 

Perhaps the release should include a hash.txt file in an archive ?

There is also perhaps an argument for having a single release file containing MOS, BBC Basic, examples and the VDP upgrade source, especially as the versions seem to be out of sync.

stevesims, 2023-10-24:
the flash tool has recently been updated and no longer requires the hash

breakintoprogram, 2023-11-11:
Going to mark this as 'Won't Do' because, as Steve said, the latest tools don't need the hash.

closed 2023-11-11 by breakintoprogram