breakintoprogram/agon-mos #101: Potential issue with stack splat in exec16 from *RUN <addr>

jackokring, 2023-10-22:
**N.B.** Is there a stack splat in `misc.asm` within `_exec16` by not setting `sp`
 to some value before `call.is`? Not for `$B0000` loading, but for say
 `$A0000` loading, as it might need a temp save, set `sp`, fix splat,
 do stuff, put stack back, `ret.lis`. It would likely be with some recusion
 and an almost at the limit 16 bit binary, assuming a high memory stack.
 Check splat in `mos.c` function `mos_cmdRUN` before `exec16`?

I mean a $B0000 is likely fine, but a potential future issue. Not really a bug yet.

jackokring, 2023-10-22:
#9 **FIX** Perhaps a `ld.sis sp, 0` or `ld.sis sp, $8000` before the `CALL.IS (DE): RET`
 based on the `A` segment being `$0B` to prevent global `(SPL)` clobber by the two byte (SPS) write of `call.is`?

This maximizes expected global size with confirmed max 32kB `*plugin` including stack, while making a logical aligned 64 kB segment except in `$B0000`. `cp $0B: jr z, ...`.

Yes I assume the `IX` to `SP` restore is a stack frame thing.

breakintoprogram, 2023-11-11:
I'll double check - the stacks * should * be set up with sensible defaults but that doesn't necessarily mean I've covered all the bases.