breakintoprogram/agon-mos #103: moscall mos_fread in Z80 mode loads only zeros

DjPoke, 2023-10-23:
Hi,
I don't know if it is a bug, but :
I'm assembling a program to load a screen from disk. And I call it from BBC BASIC, so it is ADM mode 0.
I've tested a lot of things, in Z80 mode, 'moscall mos_fread' seems to load only zeros in its buffer.

```
	.assume adl=0
	.org $F000
	.align 64
	
; equ's
fa_read: equ 01h
fa_open_existing: equ 00h

mos_fopen: equ 0Ah
mos_fclose: equ 0Bh
mos_fread: equ 1Ah
mos_cd: equ 03h

; macros
	MACRO moscall arg1
		LD A,arg1
		RST.LIS $08
    ENDMACRO

	MACRO VDU arg1
		LD A,arg1
		RST.LIS $10
    ENDMACRO

	MACRO VDU_A
		RST.LIS $10
    ENDMACRO

load_image:
	LD HL,screens_path
	moscall mos_cd
	CP 0
	JP NZ,li_error2
	
	; open the file for read
	LD HL,screen_filename
	LD C,fa_open_existing|fa_read
	moscall mos_fopen
	
	; exit on error
	CP 0
	JP Z,li_error2

	; filehandle -> C
	LD C,A

	; ignore header
	LD HL,line_buffer
	LD DE,5
	moscall mos_fread
	
	LD A,E ; DE = 5 ?
	CP 5
	JP NZ,li_error1
	LD A,D
	CP 0
	JP NZ,li_error1

	LD B,0
	
li_loop1:
	PUSH BC
	PUSH BC

	; read a line of 320 pixels
	LD HL,line_buffer
	LD DE,320
	moscall mos_fread

	POP HL ; H = y
	LD BC,$000000 ; BC = x
	
	LD A,E ; DE = 320 ?
	CP $40
	JP NZ,li_error1
	LD A,D
	CP $01
	JP NZ,li_error1

li_loop2:
	; change pixel color
	PUSH HL
	LD HL,line_buffer
	ADD HL,BC
	LD A,(HL)
	VDU 18
	VDU 0
	VDU_A
	POP HL
	
	; put a pixel to screen
	VDU 25
	VDU 69
	VDU C ; x
	VDU B
	VDU H ; y
	VDU 0
	
	INC BC
	DEC DE
	
	LD A,D
	OR E
	CP 0
	JP NZ,li_loop2
	
	POP BC
	INC B
	
	LD A,B
	CP 240
	JP NZ,li_loop1

	; close the file
	moscall mos_fclose

	LD HL,back_path
	moscall mos_cd
	CP 0
	JP NZ,li_error2

	RET
	
li_error1:
	LD HL,error1_msg
	LD BC,0
	XOR A
	RST.LIS $18
	
	; close the file
	moscall mos_fclose
	RET
	
li_error2:
	LD HL,error2_msg
	LD BC,0
	XOR A
	RST.LIS $18

	RET

error1_msg: DB "File missing !",0
error2_msg: DB "File corrupted !",0
screens_path: DB "screens",0
back_path: DB "..",0
screen_filename: DB "Screen.scn",0

```

lennart-benschop, 2023-10-23:
Long story short: this looks the same issue as #64.
mos_api_fread/fwrite assume the buffer address in HLU and length in DEU. It should conditionally call SET_AHL24 and SET_ADE24 when in Z80 mode. Note: the MSB of DEU must be set to zero, not to the current MB.

DjPoke, 2023-10-23:
So, can I set ADL to 1 in my assembly program, while BBC BASIC will call them ?
And BBC BASIC is store at $40000, no ?

DjPoke, 2023-10-23:
I've tried to reset DEU without succes.
```
	LD HL,line_buffer
	LD.lil DE,$000140
	moscall mos_fread

```

closed 2023-10-23 by DjPoke