breakintoprogram/agon-mos #116: Multi-line command history inconsistent behavior

leventp, 2023-11-12:
When pressing up-arrow a previous command is displayed as expected, again pressing up-arrow the cursor moves to the beginning of the command line.  It should display the next command in history.  Third up-arrow displays the next command in history. This behavior repeats for each command in history. Down-arrow seems to  work fine.

tonedef71, 2023-11-16:
> leaving the cursor at the beginning of the line when stepping thru history would "solve" this issue, but also feel weird in use when you want to edit a previously used command

Being able to continuously scroll up and down with the up & down keys through the command history _**unimpeded**_ is preferable.  If the "home" & "end" keys are working properly, perhaps they could be used to opt-in and/or opt-out of "command history mode" based on the current context (the cursor's current position, whether it be at the beginning, end, or middle of a line; the current keyboard mode, whether it be "command history mode" or "editing mode"; etc.).

As you mentioned, though, I'm not sure how complex the algorithms would be to implement, or whether or not folks would find using the "home" and "end" keys to influence the keyboard mode behavior to feel too _unnatural_.  I can understand why you want to transfer the command history control keys from the up & down keys to the page up & page down keys.

leventp, 2023-11-16:
Hi Steve, thanks for explaining the logic behind the design. All is well. Looking forward to '*KEY' implementation;)

stevesims, 2024-04-02:
@tonedef71  the recent Console8 2.2.x release contains support for moving thru command history using the page up and page down keys.  the up and down arrows, when at the beginning or end of the command line respectively, maintain their behaviour of also stepping thru the history.

@leventp that release also includes the `*hotkey` command