breakintoprogram/agon-mos #127: Use 0 wait states for Flash ROM

S0urceror, 2024-01-20:
According to the documentation and based on our clock speed you can run the flash with 0 wait states.
In my ElectronHAL/OS combination I have this running now for quite a while and it operates without issues.
Recommend to do this as well for MOS.

HeathenUK, 2024-01-22:
Out of curiosity, does it make a significant difference? If I think about it almost all code executed is pulled from slow storage (SD) to RAM and executed there, with relatively little code / bottleneck in flash.

That doesn't mean there's a good reason to keep it as-is, of course.

astralaster, 2024-01-22:
> According to the documentation and based on our clock speed you can run the flash with 0 wait states.

Are you sure the documentation states that? 18,4mhz is 54ns and the flash runs at 60ns. So when I'm not mistaking you are running it out of spec when using 0 waitstates. I already did some testing with 0 waitstates and it did run stable but I did not see any performance improvements.

S0urceror, 2024-01-23:
I also did the math again and you are right that 54ns is a bit slim when the RAM needs 60ns minimum. 

On the other hand how many clock cycles is the fastest instruction? For example a NOP takes 4 clock cycles so the next instruction fetch is ~200ns away. And a memory instruction takes at least 10 clock cycles. Of course the worst case is 3 consecutive NOP's that are pipelined so 3 reads each within 54ns after the other. 

Finally, the EZ80 documentation also doesn't tell if it's NOR or NAND flash. The latter has substantial shorter read times then write times. 60ns is probably the worst case.

For me 0 wait states in ElectronOS are okay but understand we want to stay on the safe side in MOS. 

As for speed benefits, it depends, inserting a 54ns wait state in a 540ns (10 clock cycle) read operation is only a bit slower.

astralaster, 2024-01-23:
https://www.mouser.com/datasheet/2/450/ps0189-26953.pdf

On page 15 they write something about flash settings.