breakintoprogram/agon-mos #130: please, improve MOS command line editing

JanB-ctrl, 2024-09-12:
Present line editing in MOS is annoying. I strongly welcome more function keys. My idea is:
left arrow   .,, moves cursor left
right arrow  ... moves cursor right
home ... moves cursor to beginning of line
end ... moves cursor to end of line
up arrow ... switch to previous line in history (with rollover)
down arrow ... switch to next line in history (with rollover)
pageup ... switch to first line in history
pagedown ... switch to last line in history
ctrl + left arrow   .,, moves cursor left one word
ctrl + right arrow  ... moves cursor right one word
ctrl + home ... delete chars from line begin to cursor
ctrl + end ... delete chars from cursor to end of line
backspace ... delete char before cursor (deleting to the left)
del ... delete char at cursor (deleting to the right)

thanks ...

stevesims, 2024-09-12:
FYI most of this functionality is already present in the Console8 variant of MOS. In spite of its name it’s compatible with all variants of the Agon Light Hardware

The latest release can be downloaded from https://github.com/AgonConsole8/agon-mos/releases/tag/v2.2.3

JanB-ctrl, 2024-09-12:
Thank You for Your answer. ;-)
I have the latest AgonLight (from OLIMEX). I was checking the version and release date of MOS and it was the same, so expect the same behavior ...
... now I looked at src code of Console8, and there are some improvements in line editing, but only a few. I was expecting that this functionality will be more standardized between MOS variants.
... Is somewhere the list of changes between MOS versions?

stevesims, 2024-09-12:
As you may have noticed the Quark firmware versions for both MOS and the VDP haven’t been updated in some time. Only Dean can merge in pull requests to the Quark repositories, and so unless he does so changes that have been made elsewhere won’t be present in the Quark versions

as there haven’t been changes to Quark for such a long time it has stopped being viable to make pull requests with the updates that the Console8 firmware has received. The MOS API in the c8 variant is essentially compatible with Quark - it contains a few new filing system API calls exposing some fatfs functionality which a few command line tools rely on. The CLI supports several new commands, which includes function key support 

the c8 VDP contains extensive changes. This includes full support for Acorn’s basic GCOL operations (including those added to RISC OS), more PLOT operations, a font management API, graphics context API, and a few more things. Programs written to work with the Quark 1.04 VDP will work fine on the latest Console8 VDP release, except for a few minor cases that rely on buggy behaviour in the Quark 1.04 VDP that have been fixed in the c8 variant.

Info on firmware features including which releases gained support for new APIs can be found on the agon-docs community documentation site.  https://agonconsole8.github.io/agon-docs/

JanB-ctrl, 2024-09-12:
... so do You suggest to move all SW to Console8 versions? Is it possible/safe (with respect to HW diferencies)?

stevesims, 2024-09-12:
The Console8 firmware is built to run on all Agon Light variants. There are no disadvantages or risks involved in running it on an an Olimex Agon Light 2 - only advantages. It is completely safe and extensively tested

Olimex themselves have started installing it on machines - however they have a lot of stock with older versions of firmware installed and don’t update their hardware before sending it out to customers

JanB-ctrl, 2024-09-12:
OK. 
So I update my OlimexAgonLight2 and move to Console8 variant... ;-)
BTW: are there solved some another "boring" issues,  like:
- absence of informative prompt in comand line (showing at least directory or date/time. Like settings "PROMPT $p$g - $t" in normal DOS)?
- ability to better control output of DIR command (I mean using wildcards, paging of longer listings, or widening listings). Like a "DIR abd*", "DIR /p" or "DIR /w" or "DIR /on")?

thanks a lot,
JB

stevesims, 2024-09-12:
The MOS command prompt on the C8 version shows your current directory. It is not currently programmable tho. I have ideas about adding that functionality in, but haven’t started to work on it yet. Those ideas are documented in issues on the console8 agon-mos repository 

“Paged mode” can be turned on by pressing ctrl-N on either Quark or c8 firmware. The paging behaviour is crude - inherited from the Acorn BBC Micro - you press the Shift key to allow the next page of text to appear. CTRL-O will clear this behaviour. This will work for directory listings, and other places too such as listing a program in BASIC

Directory listings on the C8 MOS will show directories in a different colour, are alphabetically sorted (directories first) and by default shown in a multi-column display. It supports a single flag to show “long” mode. Wildcards are supported. Running programs directly using their filename is also supported, as is tab-completion

JanB-ctrl, 2024-09-12:
Sounds good. I will try it ... ;-)

Thanks,
JB

stevesims, 2025-04-30:
@JanB-ctrl FYI MOS 3 was released recently.  That includes a whole raft of new features, amongst which are the ability to define your command prompt, improvements to the `Dir` command, and optional automatic use of paged output for commands that can produce long output, such as the `Dir` command.